OSCPU/NutShell #1: How to add a new AM riscv64-nutshell in the AM project?

zkh2018, 2020-07-08:


sashimi-yzh, 2020-07-10:
Hi, we are still refining the code. We will make other repos available to the public in a couple of days.

We have already [updated README.md](https://github.com/OSCPU/NutShell/commit/370701d3bffa269b1b12f0f3535feb31276c4bb5) to show how to run the built-in images. Please try it first.

zkh2018, 2020-07-13:
Hi sashimi-yzh,

I follow your updated readme: 
```
make IMAGE=./ready-to-run/linux.bin emu
```
But the simulator didn't work:
```
[src/device/io/mmio.c,13,add_mmio_map] Add mmio map 'sdhci' at [0xa3000000, 0xa300007f]
[src/device/sdcard.c,121,init_sdcard] Can not find sdcard image: /home/yzh/projectn/debian.img
[src/device/io/mmio.c,13,add_mmio_map] Add mmio map 'difftest.serial' at [0xa10003f8, 0xa10003ff]
bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
      .....
       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.18.0-ga2ee7cb2-dirty (wkf@9900k) (gcc version 9.2.0 (GCC)) #973 Mon Jul 6 18:41:57 CST 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (4417024 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Cannot allocate SWIOTLB buffer
[    0.000000] elf_hwcap is 0x112d
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 7575
[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon
[    0.000000] Dentry cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 24852K/30720K available (666K kernel code, 80K rwdata, 102K rodata, 4376K init, 98K bss, 5868K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.000000] pid_max: default: 4096 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.040000] Unpacking initramfs...
[    0.090000] workingset: timestamp_bits=62 max_order=13 bucket_order=0
[    0.100000] random: get_random_bytes called from 0xffffffff8044998a with crng_init=0
[    0.110000] Freeing unused kernel memory: 4376K
[    0.110000] This architecture does not have kernel memory protection.
now = 171s
now = 231s
mount: mounting proc on /proc failed: No such device
```
The log stays here and doesn't go down.

### The another question :
In line 41558 TopMain.v file, there is no definition of RAMHelper module. We also noticed that it is defined in NutShell/src/test/vsrc/ram.v, but we need package DPI-C (import "DPI-C" function void ram_helper), which is not found in the project. How to solve this problem? 
Thanks
```
  RAMHelper RAMHelper ( // @[AXI4RAM.scala 36:21]
    .clk(RAMHelper_clk),
    .rIdx(RAMHelper_rIdx),
    .rdata(RAMHelper_rdata),
    .wIdx(RAMHelper_wIdx),
    .wdata(RAMHelper_wdata),
    .wmask(RAMHelper_wmask),
    .wen(RAMHelper_wen)
  );
```

wakafa1, 2020-07-13:
Hi, here are the answers to your questions.

1. What you see is the log printed by the Linux kernel, showing the boot process successfully. Since the speed of verilog simulation is not so fast, you may wait for up to 10-20 min to get BusyBox launched and see its output like `/# `. By the way, we generated the linux.bin specially to make Redis run first so there will be some log printed by Redis too.

2. There're two things involved here. Firstly, src/test/vsrc/ram.v has already been included as simulation file:
https://github.com/OSCPU/NutShell/blob/370701d3bffa269b1b12f0f3535feb31276c4bb5/Makefile#L56
Secondly, DPI-C is a standard interface which can be used to interface SystemVerilog with C. Verilator(our simulator) supports this kind of language feature.

sashimi-yzh, 2020-07-26:
HI, @zkh2018 . 

We have released the [AM project](https://github.com/OSCPU/nexus-am) and the project already contains `riscv64-nutshell`. So you do not have to add a new AM by yourself.

Please let us know whether you have any other question.

sashimi-yzh, 2020-07-28:
Hi, @zkh2018 .

It seems that you do not have other questions about this issue. I am going to close this discussion.

Welcome to report new issues. Thanks!

closed 2020-07-28 by sashimi-yzh