OSCPU/NutShell #5: CPU hangs up when accessing memory just below MMIO address space

imhcyx, 2020-07-14:
I'm using the following settings on my FPGA board:
```
object MyFPGASettings {
  def apply() = Map(
    "NrExtIntr" -> 3,
    "ResetVector" -> 0x10000000L,
    "MMIOBase" -> 0x0000000080000000L,
    "MMIOSize" -> 0x0000000080000000L
  )
}
```
This makes 0x80000000-0xffffffff MMIO address space and I can use 0x40000000-0x7fffffff as memory address space.

However, when I try to access memory at 0x7fffffc0, the system hangs up . I wonder if this memory access will trigger Prefetcher to prefetch memory at 0x80000000 (but actually in MMIO address space) and thus hang up the entire system?

ZLJ987, 2020-07-28:
Hi imhcyx,
this bug is probably because access request at 0x7fffffc0 in L2 CacheStage3 was blocked and accidently flushed by the prefetch request in Stage2. Now we have fixed this bug in prefetcher so L2 Cache will not be accessed if the prefetch request is mapped into MMIO space in branch "bug-prefetch-mmio". Please let us know if works in your FPGA board and thanks for pointing out the problem!

imhcyx, 2020-07-28:
@ZLJ987 
Seems the problem still exists...

sashimi-yzh, 2020-07-28:
@imhcyx Would you please have a try with the prefetcher disabled?

imhcyx, 2020-07-28:
> @imhcyx Would you please have a try with the prefetcher disabled?

I've tried disabling the prefetcher and L2 cache before applying the patch above and no problem has occured.

sashimi-yzh, 2020-07-28:
@imhcyx What about only disabling the prefetcher but keeping the L2 cache enabled?

sashimi-yzh, 2020-07-31:
Fixed in #21

closed 2020-07-31 by sashimi-yzh