OSCPU/NutShell #8: where is Noop ?

MasterJerryZh, 2020-07-23:
Is Noop a open source project?
If it is, where is it? 
I cannot find any repo about it, could you please show me the url of Noop repo to me? 
Thanks!

sashimi-yzh, 2020-07-23:
NOOP is designed as a programming assignment for educational purpose. It is still not stable and may be redesigned every year. Therefore there is no official release yet.

But here are the reference implementation:
* Code in this NutShell repo before 2019/08/27 can be considered as NOOP designed in 2019.
* [Here](https://github.com/nju-mips/noop-lo) is an implementation of NOOP designed by students in NJU at 2018. This is the CPU design which won the 2nd prize in LoongsonCup18. It is designed with Chisel, too. But note that the ISA is mips32, since the contest is held by Loongson. Also the implementation is very different from the one above.

sashimi-yzh, 2020-07-25:
We have added your question to README.md. Thank you for the feedback!

closed 2020-07-25 by sashimi-yzh