OSCPU/NutShell #10: difftest.cpp:40: void init_difftest(rtlreg_t*): Assertion `handle' failed.

iam41xin, 2020-07-27:
Hi mates,

While I run the simulation by 'make emu', I met 2 issues:

make[1]: Leaving directory '/NutShell/build/emu-compile'
Using seed = 7348
The image is ready-to-run/linux.bin
[warning] sdcard img not found
emu: /NutShell/src/test/csrc/difftest.cpp:40: void init_difftest(rtlreg_t*): Assertion `handle' failed.
make: *** [Makefile:100: emu] Aborted

As above shown, there is a warning that image files were not found, this issue has been solved after I modified the sdcard.cpp in folder $NUTSHELL_HOME/src/test/csrc/.
But, the 2nd issue still couldn't be solved, in Makefile, default REF_SO has been setup correctly, why assertion 'handle' still failed?

Thanks!

wakafa1, 2020-07-27:
Usually mismatched NEMU can cause such error. I just cloned and built this project from scratch but cannot reproduce the problem you mentioned. Are you using the ready-to-run mode? Or did you compile a copy of nemu yourself?

iam41xin, 2020-07-28:
Hi Wakafa,
Initially, what I've done is nothing but to clone the git, download and install mill and run make && make emu in folder $NUTSHELL_HOME.
I've tried to set $(USE_READY_TO_RUN_NEMU) option from true to false and clone nemu, after setting NEMU_HOME, 'make emu' compiled the file riscv64-nemu-interpreter-so successfully in NEMU_HOME/build, but still, Assertion `handle' failed.
Also I compiled microbench-riscv64-nutshell.bin with nexus-am successfully, using this file to replace linux.bin, I got the same result, Assertion `handle' failed.
My OS is Linux CentOS 7, x86_64.

iam41xin, 2020-07-28:
I change the gcc version from 4.8.5(year 2015) to 8.3.1(year 2018) and leave the option $(USE_READY_TO_RUN_NEMU) as false, it works now.
Thanks and I'll close this issue.

closed 2020-07-28 by iam41xin