OSCPU/NutShell #14: make emu时emu: NutShell/src/test/csrc/difftest.cpp:40: void init_difftest(rtlreg_t*): Assertion `handle' failed.

Aerotic, 2020-07-29:
根据文档安装依赖后，make emu会报出如下信息
```shell
CPPFLAGS=-DREF_SO=\\\"./ready-to-run/riscv64-nemu-interpreter-so\\\" make -C ./build/emu-compile/ -f /home/rv/riscv/NutShell/build/emu-compile/VNutShellSimTop.mk
make[1]: Entering directory '/home/rv/riscv/NutShell/build/emu-compile'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/home/rv/riscv/NutShell/build/emu-compile'
Using seed = 2523
The image is ready-to-run/linux.bin
[warning] sdcard img not found
emu: /home/rv/riscv/NutShell/src/test/csrc/difftest.cpp:40: void init_difftest(rtlreg_t*): Assertion `handle' failed.
make: *** [Makefile:102: emu] Aborted (core dumped)
```

sashimi-yzh, 2020-07-29:
请尝试进行如下修改, 然后提供`perror`函数输出的报错信息, 谢谢!
```diff
diff --git a/src/test/csrc/difftest.cpp b/src/test/csrc/difftest.cpp
index 5f5aff9..5c9a42b 100644
--- a/src/test/csrc/difftest.cpp
+++ b/src/test/csrc/difftest.cpp
@@ -37,6 +37,9 @@ void difftest_skip_dut() {
 void init_difftest(rtlreg_t *reg) {
   void *handle;
   handle = dlopen(REF_SO, RTLD_LAZY | RTLD_DEEPBIND);
+  if (handle == NULL) {
+    perror("Error when opening " REF_SO);
+  }
   assert(handle);

   ref_difftest_memcpy_from_dut = (void (*)(paddr_t, void *, size_t))dlsym(handle, "difftest_memcpy_from_dut");
```

Aerotic, 2020-07-30:
> ```diff
> +  if (handle == NULL) {
> +    perror("Error when opening " REF_SO);
> +  }
> ```

```shell
[warning] sdcard img not found
Error when opening ./ready-to-run/riscv64-nemu-interpreter-so: No such file or directory
emu: /home/rv/riscv/NutShell/src/test/csrc/difftest.cpp:50: void init_difftest(rtlreg_t*): Assertion `handle' failed.
make: *** [Makefile:102: emu] Aborted (core dumped)
```

sashimi-yzh, 2020-07-30:
我们这边在Debian10 和Ubuntu 18.04中都无法复现这个错误, 能否请你帮忙提供一下你的系统信息?

Aerotic, 2020-07-30:
> 我们这边在Debian10 和Ubuntu 18.04中都无法复现这个错误, 能否请你帮忙提供一下你的系统信息?

ubuntu20.04

wakafa1, 2020-07-30:
能否请你提供一下你的 GCC 版本？我注意到之前 #10 提到了类似的问题，是 GCC 版本过低导致的。

Aerotic, 2020-07-30:
> 能否请你提供一下你的 GCC 版本？我注意到之前 #10 提到了类似的问题，是 GCC 版本过低导致的。

rv@rv:~/riscv/NutShell$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.3.0-10ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)

Aerotic, 2020-07-30:
> 能否请你提供一下你的 GCC 版本？我注意到之前 #10 提到了类似的问题，是 GCC 版本过低导致的。

rv@rv:~/riscv/NutShell$ riscv64-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc-cross/riscv64-linux-gnu/9/lto-wrapper
Target: riscv64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.3.0-10ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libsanitizer --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --without-target-system-zlib --enable-libpth-m2 --enable-multiarch --disable-werror --disable-multilib --with-arch=rv64imafdc --with-abi=lp64d --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=riscv64-linux-gnu --program-prefix=riscv64-linux-gnu- --includedir=/usr/riscv64-linux-gnu/include
Thread model: posix
gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu1)

wakafa1, 2020-07-30:
@Aerotic  我在 Ubuntu 20.04 LTS 环境下成功复现出了这一情况，初步考虑是运行环境的问题，后续有待完善。
不过我发现手动编译 NEMU 可以解决这一报错，你可以尝试一下：
1. 设置 Makefile 中 USE_READY_TO_RUN_NEMU=false
2. 克隆 https://github.com/OSCPU/nemu
3. 设置 NEMU_HOME 环境变量为 nemu 所在位置
4. 在 NutShell 工程中 make clean，然后重新 make emu

sashimi-yzh, 2020-07-30:
@Aerotic 我们更新了so文件(请见#16) , 在ubuntu 20.04下可以使用, 也请你试一下是否能使用.

感谢帮助我们发现问题.

Aerotic, 2020-07-30:
> @Aerotic 我们更新了so文件(请见#16) , 在ubuntu 20.04下可以使用, 也请你试一下是否能使用.
> 
> 感谢帮助我们发现问题.

也谢谢你们的快速回复 ：）

Aerotic, 2020-07-30:
> @Aerotic 我们更新了so文件(请见#16) , 在ubuntu 20.04下可以使用, 也请你试一下是否能使用.
> 
> 感谢帮助我们发现问题.

我试了一下，
在Makefile 中 USE_READY_TO_RUN_NEMU=true的情况下，make clean然后make emu
可以执行到如下这个地步了
```shell
bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 4.18.0-ga2ee7cb2-dirty (wkf@9900k) (gcc version 9.2.0 (GCC)) #973 Mon Jul 6 18:41:57 CST 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (4417024 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000081ffffff]
[    0.000000] Cannot allocate SWIOTLB buffer
[    0.000000] elf_hwcap is 0x112d
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 7575
[    0.000000] Kernel command line: root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon
[    0.000000] Dentry cache hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 24852K/30720K available (666K kernel code, 80K rwdata, 102K rodata, 4376K init, 98K bss, 5868K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000000] console [hvc0] enabled
[    0.000000] console [hvc0] enabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] bootconsole [early0] disabled
[    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.000000] pid_max: default: 4096 minimum: 301
[    0.000000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
[    0.000000] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.000000] clocksource: Switched to clocksource riscv_clocksource
[    0.040000] Unpacking initramfs...
[    0.090000] workingset: timestamp_bits=62 max_order=13 bucket_order=0
[    0.100000] random: get_random_bytes called from 0xffffffff8044998a with crng_init=0
[    0.110000] Freeing unused kernel memory: 4376K
[    0.110000] This architecture does not have kernel memory protection.
now = 124s
now = 184s
mount: mounting proc on /proc failed: No such device
now = 245s
```

AugustusWillisWang, 2020-07-30:
> 我试了一下，git pull之后
> 在Makefile 中 USE_READY_TO_RUN_NEMU=true的情况下，make clean然后make emu
> 可以执行到如下这个地步了

```
a5 different at pc = 0xffffffff80485040, right= 0x0000000020207093, wrong = 0x00000000202070d3
ABORT at pc = 0x7f80485042
total guest instructions = 135313281
```

这个问题是由于处理器在推测执行过程中会推测性地更新页表项中的access和dirty, 而模拟器NEMU则不会. 这样, 在涉及页表项中的access和dirty时, 处理器与模拟器的行为不一致.

后续我们会修改处理器代码与模拟器代码, 在仿真中默认关闭页表项中access和dirty的更新.

wakafa1, 2020-07-30:
@Aerotic 你好，这个问题已经被 #17 修复了，非常感谢你的反馈！

sashimi-yzh, 2020-07-30:
@Aerotic 请问问题是否已解决?

Aerotic, 2020-07-30:
已解决，谢谢😄------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;Zihao&amp;nbsp;Yu&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年7月30日(星期四) 晚上11:06
收件人:&nbsp;&quot;OSCPU/NutShell&quot;<NutShell@noreply.github.com&gt;;
抄送:&nbsp;&quot;Aero&quot;<654868961@qq.com&gt;;&quot;Mention&quot;<mention@noreply.github.com&gt;;
主题:&nbsp;Re: [OSCPU/NutShell] make emu时emu: NutShell/src/test/csrc/difftest.cpp:40: void init_difftest(rtlreg_t*): Assertion `handle&#39; failed. (#14)

closed 2020-07-30 by sashimi-yzh

trueif, 2021-12-30:
可以试试香山的ready-to-run。里面的linux.bin可以起出"Hello, RISC-V World!"字样。

https://github.com/OpenXiangShan/ready-to-run/tree/66a5be63980803e3959497fd5b1a48f54a551971