OSCPU/NutShell #22: 按照教程制作SD卡镜像后，插入PYNQ-Z2板子上跑无反应，且板载FTDI那个串口无输出

Aerotic, 2020-08-01:
pynq的镜像是可以跑的，因此板子没有问题，能否提供示例的BOOT.BIN Image system.dtb ？

wakafa1, 2020-08-01:
@Aerotic  你好，我们昨天上传了一个针对 pynq 的[运行文档](https://github.com/OSCPU/NutShell/blob/master/fpga/boot/standalone.md)。不知道是否与你制作SD卡镜像的操作有一些出入，如果不一样的话请重新制作一个与 pynq 相匹配的 BOOT.BIN 再尝试一下。

Aerotic, 2020-08-01:
> @Aerotic 你好，我们昨天上传了一个针对 pynq 的[运行文档](https://github.com/OSCPU/NutShell/blob/master/fpga/boot/standalone.md)。不知道是否与你制作SD卡镜像的操作有一些出入，如果不一样的话请重新制作一个与 pynq 相匹配的 BOOT.BIN 再尝试一下。

你们发的运行文档有点问题：
1. SW0 这个开关，我往上拨动到LD4一侧才能让串口打印出来东西
2. 目前给出的RV_BOOT.BIN https://github.com/OSCPU/NutShell/blob/master/fpga/boot/pynq/RV_BOOT.BIN 这个链接失效了

目前，我已运行到如下地步
```shell
[19:32:52:537] RISC-V Boot Environment Setup... ␍␊
[19:32:52:537] open...␍␊
[19:32:52:569] SD Polled RISC-V boot file failed ␍␊
```

wakafa1, 2020-08-01:
你好，请使用 NutShell/fpga/boot/pynq/RV_BOOT.BIN

Aerotic, 2020-08-01:
> 你好，串口信息提示未找到 RV_BOOT.bin 镜像，请使用 NutShell/fpga/boot/pynq/RV_BOOT.bin

已运行到如下界面：
```shell
[20:53:26:538] RISC-V Boot Environment Setup... ␍␊
[20:53:26:538] open...␍␊
[20:53:26:569] lseek...␍␊
[20:53:26:569] read...␍␊
[20:53:28:362] close...␍␊
[20:53:28:362] Successfully load RISC-V boot file into DRAM @ 0x10000000 ␍␊
[20:53:28:362] Reset RISC-V core.␍␊
[20:53:28:362] bbl loader␍␊
[20:53:28:362] freq-mhz = 60␊
[20:53:28:362] CLINT: set frequency to 60 MHz␊
[20:53:28:362]               vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:378]                   vvvvvvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:378] rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:378] rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:378] rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:394] rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:394] rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv␍␊
[20:53:28:394] rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  ␍␊
[20:53:28:394] rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    ␍␊
[20:53:28:410] rr                vvvvvvvvvvvvvvvvvvvvvv      ␍␊
[20:53:28:410] rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr␍␊
[20:53:28:410] rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr␍␊
[20:53:28:410] rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr␍␊
[20:53:28:425] rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr␍␊
[20:53:28:425] rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr␍␊
[20:53:28:425] rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr␍␊
[20:53:28:442] rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr␍␊
[20:53:28:442] rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr␍␊
[20:53:28:442] rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr␍␊
[20:53:28:442] rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr␍␊
[20:53:28:458] rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr␍␊
[20:53:28:458] ␍␊
[20:53:28:458]        INSTRUCTION SETS WANT TO BE FREE␍␊
[20:53:28:490] [    0.000000] OF: fdt: Ignoring memory range 0x60000000 - 0x60200000␍␊
[20:53:28:490] [    0.000000] Linux version 4.18.0-gc4490f2d (wkf@9900k) (gcc version 9.2.0 (GCC)) #978 Fri Jul 31 09:17:43 CST 2020␍␊
[20:53:28:505] [    0.000000] bootconsole [early0] enabled␍␊
[20:53:28:505] [    0.000000] earlycon: earlycon_map: Couldn't map 0x00000000e0000000␍␊
[20:53:28:522] [    0.000000] earlycon: cdns0 at MMIO 0x00000000e0000000 (options '115200n8')␍␊
[20:53:28:522] [    0.000000] Zone ranges:␍␊
[20:53:28:522] [    0.000000]   DMA32    empty␍␊
[20:53:28:522] [    0.000000]   Normal   [mem 0x0000000060200000-0x000000006fffffff]␍␊
[20:53:28:538] [    0.000000] Movable zone start for each node␍␊
[20:53:28:538] [    0.000000] Early memory node ranges␍␊
[20:53:28:538] [    0.000000]   node   0: [mem 0x0000000060200000-0x000000006fffffff]␍␊
[20:53:28:553] [    0.000000] Initmem setup node 0 [mem 0x0000000060200000-0x000000006fffffff]␍␊
[20:53:31:481] [    0.000000] software IO TLB [mem 0x6bc7d000-0x6fc7d000] (64MB) mapped at [(____ptrval____)-(____ptrval____)]␍␊
[20:53:31:498] [    0.000000] elf_hwcap is 0x112d␍␊
[20:53:31:498] [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64135␍␊
[20:53:31:514] [    0.000000] Kernel command line: console=ttyPS0,115200n8 root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait earlycon clk_ignore_unused␍␊
[20:53:31:529] [    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes)␍␊
[20:53:31:529] [    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes)␍␊
[20:53:31:546] [    0.000000] Sorting __ex_table...␍␊
[20:53:31:770] [    0.000000] Memory: 183724K/260096K available (3917K kernel code, 249K rwdata, 1288K rodata, 140K init, 790K bss, 76372K reserved, 0K cma-reserved)␍␊
[20:53:31:786] [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1␍␊
[20:53:31:786] [    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0␍␊
[20:53:31:786] [    0.000000] slcr mapped to (____ptrval____)␍␊
[20:53:31:801] [    0.000000] plic: mapped 3 interrupts to 1 (out of 1) handlers.␍␊
[20:53:31:817] [    0.000000] zynq_clock_init: clkc starts at (____ptrval____)␍␊
[20:53:31:817] [    0.000000] Zynq clock init␍␊
[20:53:31:866] [    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns␍␊
[20:53:31:882] [    0.000000] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)␍␊
[20:53:31:882] [    0.000000] pid_max: default: 32768 minimum: 301␍␊
[20:53:31:897] [    0.020000] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)␍␊
[20:53:31:897] [    0.020000] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)␍␊
[20:53:31:945] [    0.070000] devtmpfs: initialized␍␊
[20:53:31:961] [    0.090000] random: get_random_u32 called from bucket_table_alloc+0x70/0x10a with crng_init=0␍␊
[... 46 lines cut ...]
[20:53:34:106] [    2.190000] sdhci: Copyright(c) Pierre Ossman␍␊
[20:53:34:106] [    2.190000] sdhci-pltfm: SDHCI platform and OF driver helper␍␊
[20:53:34:154] [    2.250000] mmc0: SDHCI controller on e0100000.mmc [e0100000.mmc] using ADMA␍␊
[20:53:34:175] [    2.260000] usbcore: registered new interface driver usbhid␍␊
[20:53:34:175] [    2.260000] usbhid: USB HID core driver␍␊
[20:53:34:203] [    2.290000] NET: Registered protocol family 10␍␊
[20:53:34:234] [    2.320000] Segment Routing with IPv6␍␊
[20:53:34:234] [    2.330000] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver␍␊
[... cut ...]
[20:53:34:603] [    2.690000] EXT4-fs (mmcblk0p2): recovery complete␍␊
[20:53:34:618] [    2.700000] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)␍␊
[20:53:34:618] [    2.710000] VFS: Mounted root (ext4 filesystem) on device 179:2.␍␊
[20:53:34:634] [    2.730000] devtmpfs: mounted␍␊
[20:53:34:651] [    2.740000] Freeing unused kernel memory: 140K␍␊
[20:53:34:651] [    2.740000] This architecture does not have kernel memory protection.␍␊

```

请问网口的ip是固定在RV_BOOT.bin里的么 可以手动调么

wakafa1, 2020-08-01:
RV_BOOT.bin 只是一个简单的 BBL+Linux Kernel 镜像，不包含 IP 信息；IP 地址可以通过修改 Debian 的 /etc/network/ 下相关文件进行配置。
另外我注意到你贴出来的串口信息仅是 Linux Kernel 的启动日志，后续有无 Debian 的启动信息？如果没有建议检查一下 SD 卡 rootfs 的配置情况，需要注意 fpga/boot/README.md 里面介绍的步骤是针对 PS（arm核）的，而在 Stand-alone（pynq）模式下我们的 rootfs 架构需要是 riscv64 的。

Aerotic, 2020-08-02:
> RV_BOOT.bin 只是一个简单的 BBL+Linux Kernel 镜像，不包含 IP 信息；IP 地址可以通过修改 Debian 的 /etc/network/ 下相关文件进行配置。
> 另外我注意到你贴出来的串口信息仅是 Linux Kernel 的启动日志，后续有无 Debian 的启动信息？如果没有建议检查一下 SD 卡 rootfs 的配置情况，需要注意 fpga/boot/README.md 里面介绍的步骤是针对 PS（arm核）的，而在 Stand-alone（pynq）模式下我们的 rootfs 架构需要是 riscv64 的。

请问 Stand-alone（pynq）模式下我们的 rootfs 架构如何生成呢  我发现debian好像没有riscv版的

wakafa1, 2020-08-02:
你好，请参考 Debian-riscv 官网 https://wiki.debian.org/RISC-V 中 “Creating a riscv64 chroot” 这一章节里的内容（需要把debootstrap 替换为 qemu-debootstrap）

Aerotic, 2020-08-02:
> 你好，请参考 Debian-riscv 官网 https://wiki.debian.org/RISC-V#debootstrap 中的内容，具体指令应该是 `sudo qemu-debootstrap --arch riscv64 unstable /mnt http://deb.debian.org/debian-ports`

已从串口成功启动
```shell
[19:07:39:830] root@rv:~# cat /proc/cpuinfo␍␊
[19:07:51:526] hart⇥	: 0␍␊
[19:07:51:526] isa⇥	: rv64imafdc␍␊
[19:07:51:526] mmu⇥	: sv39␍␊
[19:07:51:526] uarch⇥	: UCAS,NutShell 1.0␍␊
```

wakafa1, 2020-08-02:
好的，感谢你的反馈，我们也会在后续不断完善 FPGA 相关文档。

closed 2020-08-02 by wakafa1