OSCPU/NutShell #36: make emu error

zhoupeng, 2020-08-25:
***** What happened *****
$ make emu
mkdir -p build
mill chiselModule.test.runMain top.TopMain -td build --output-file NutShellSimTop.v BOARD=sim CORE=seq  
Preparing Java 14-ea runtime; this may take a minute or two ...
Compiling ~/OSCPU/NutShell.git/build.sc
build.sc:5: value repositories is not a member of mill.Module with mill.scalalib.ZincWorkerModule
  def repositories() = super.repositories ++ Seq(
                             ^
build.sc:5: `T.sources` definitions must have 0 parameter lists
  def repositories() = super.repositories ++ Seq(
      ^
build.sc:4: illegal inheritance;
 self-type build.this.CustomZincWorkerModule.type does not conform to mill.scalalib.ZincWorkerModule's selftype mill.scalalib.ZincWorkerModule with mill.scalalib.CoursierModule
object CustomZincWorkerModule extends ZincWorkerModule {
                                      ^
Compilation Failed

**** Software environment *****
$ java --version
openjdk 14-ea 2020-03-17
OpenJDK Runtime Environment (build 14-ea+18-Ubuntu-1)
OpenJDK 64-Bit Server VM (build 14-ea+18-Ubuntu-1, mixed mode, sharing)

$ mill --version
Mill Build Tool version 0.8.0
Java version: 14-ea, vendor: Private Build, runtime: /usr/lib/jvm/java-14-openjdk-amd64
Default locale: en_US, platform encoding: UTF-8
OS name: "Linux", version: 5.0.0-38-generic, arch: amd64

$ verilator --version
Verilator 4.016 2019-06-16 rev UNKNOWN_REV

$ gcc --version
gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

wakafa1, 2020-08-25:
Hi @zhoupeng ,
I found that the latest version of mill (0.8.0) triggers this problem. Maybe the interface has changed and we will check it out later.
Please use mill 0.7.4 instead (refer to [this link](https://www.lihaoyi.com/mill/index.html#manual)). If you have other problems, please feel free to tell us.

zhoupeng, 2020-08-26:
mill 0.7.4 works, 
Thanks

closed 2020-08-26 by zhoupeng