OSCPU/NutShell #51: 能否提供Golden的TopMain.v文件? Vivado BD创建失败, 因为NutShell模块的port不被识别为io_mem AXI口

huangping6686, 2021-05-25:
您好!
能否提供正确的TopMain.v文件,供参考?  我使用0.6.3的mill创建了TopMain.v文件, 在fpga目录下, 使用pynq板卡, 创建vivado工程时,得到错误: 
########## Vivado ERROR LOG ########
ERROR: [BD 5-106] Arguments to the connect_bd_intf_net command cannot be empty.
ERROR: [Common 17-39] 'connect_bd_intf_net' failed due to earlier errors.

    while executing
"connect_bd_intf_net -intf_net NutShell_0_io_mmio [get_bd_intf_pins io_mmio] [get_bd_intf_pins NutShell_0/io_mmio]"

原因是io_mmio_* 接口不被识别为 BD的AXI interface;   
[image]

谢谢!
Ping

AugustusWillisWang, 2021-05-25:
您好! 这一现象出现的原因是没有生成面向 FPGA 的 verilog 代码, 而是面向仿真环境的 verilog.

请检查生成 verilog 时使用的参数. 以 pynq FPGA 为例, 应使用的命令是:
```sh
make verilog BOARD=pynq
```

NutShell的顶层接口定义在[NutShell.scala](https://github.com/OSCPU/NutShell/blob/master/src/main/scala/system/NutShell.scala), 您可以参考这一文件来获取更多细节.

huangping6686, 2021-05-26:
感谢您的迅速答复;  用 make verilog BOARD=pynq, 已经解决了问题;    祝好!

closed 2021-05-26 by AugustusWillisWang