OSCPU/NutShell #52: 怎么跑一个简单的code并生成wf呢

lenged, 2021-06-22:
marobench 里面包含好多个测试程序，能否写一段程序并进行测试呢？

wakafa1, 2021-06-23:
你好，我的理解是你想要写一个自己的测试程序在果壳上运行起来.
我们项目中有一个裸机运行时环境（AM）用于生成 workload，microbench 是其中内置的一个，你也可以写一个与 microbench 并列的 APP 生成 workload.
关于 AM 的内容请参考本项目 https://github.com/OSCPU/nexus-am 以及南京大学 PA 课程中的相关讲解 https://nju-projectn.github.io/ics-pa-gitbook/ics2020/2.3.html

lenged, 2021-06-26:
谢谢， 我试下

lenged, 2021-07-03:
执行过程中遇到1个问题， 
make ARCH=riscv64-nemu  貌似check 不过， AM repo 只提供了am/arch/riscv64-nutshell.mk?

lenged, 2021-07-04:
hi,  wakafa１
自己写了一个hello world 编译出现如下问题, 另外请教下NEMU/AM　是否可以调用glibc ? 
``
riscv64-linux-gnu-ld: project/riscv/NutShell_top/nexus-am/am/build/am-riscv64-nutshell.a(trm.o): in function `_trm_init':
project/riscv/NutShell_top/nexus-am/am/src/nutshell/isa/riscv/trm.c:33: undefined reference to `main'
make: *** [project/riscv/NutShell_top/nexus-am/am/arch/riscv64-nutshell.mk:26: image] Error 1
``
in trm.c
``
int main(const char *args)
``
in my test.cpp
``
int main(const char* args) {
    using namespace std;
    cout << "hello world\n";
}
``

wakafa1, 2021-07-05:
> 执行过程中遇到1个问题，
> make ARCH=riscv64-nemu 貌似check 不过， AM repo 只提供了am/arch/riscv64-nutshell.mk?

可以参考一下香山项目下的 AM，它是完整版的：https://github.com/OpenXiangShan

wakafa1, 2021-07-05:
> hi, wakafa１
> 自己写了一个hello world 编译出现如下问题, 另外请教下NEMU/AM　是否可以调用glibc ?
> `` riscv64-linux-gnu-ld: project/riscv/NutShell_top/nexus-am/am/build/am-riscv64-nutshell.a(trm.o): in function `_trm_init': [...]

AM 不能调用 glibc，不过 AM 有一个内置的 klib，包含有一些常用的输入输出、字符串函数等；想用 glibc 的话，建议编进 ramfs 在 Linux 环境下跑

lenged, 2021-07-11:
inlcude klib 可以顺利跑起来了， 谢谢

closed 2021-07-11 by lenged