OSCPU/NutShell #54: vcd dump 的waveform 中clk 始终没有toggle

lenged, 2021-07-18:
请教一个问题,  vcd waveform 中， 所有的clock 都是tie=1； 但是看pc 是递增的， 这个是正常情况吗？ 目前还不知道git 怎么图片...

lenged, 2021-07-18:
add wf
[image]

wakafa1, 2021-07-18:
这个看起来不正常，clock 信号应该是会有翻转的；这是 NutShell master 分支出现的情况吗？

poemonsense, 2021-07-19:
这是正常的，因为NutShell只会在clock为1的时候打波形，为了提高仿真效率。

具体代码参考 src/test/csrc/emu.h 的 95 行开始的 single_cycle() 函数，https://github.com/OSCPU/NutShell/blob/master/src/test/csrc/emu.h#L95

lenged, 2021-07-25:
@poemonsense 
对verilator dump wf 不是很了解， 看了文档，Nutshell 目前的tb clock 不是按照time 来步进， 是按照clock  cycle 来步进的？

我改成如下code； 可以正常toggle了， 仿真从log 来看没有区别；不知道cycles 改变会不会影响 正常仿真

```c++
 void single_cycle() {
    dut_ptr->clock = 0;
    dut_ptr->eval();

#if VM_TRACE
    tfp->dump(cycles);
#endif
    cycles ++;

    dut_ptr->clock = 1;
    dut_ptr->eval();

#if VM_TRACE
    tfp->dump(cycles);
#endif

    cycles ++;

  }
```
[image]

poemonsense, 2021-07-25:
> Nutshell 目前的tb clock 不是按照time 来步进， 是按照clock cycle 来步进的？

不是，现在的仿真中如果time的单位是1，clock cycle其实是2。verilator里面每次eval应该是一个时间单位（不过这点我不是很确定）。之所以之前的波形里面clock没有变化，是因为只在clock为1的时候输出了波形，clock为0的时候没有输出

> 我改成如下code； 可以正常toggle了， 仿真从log 来看没有区别；不知道cycles 改变会不会影响 正常仿真
>
> ```c++
>  void single_cycle() {
>     dut_ptr->clock = 0;
>     dut_ptr->eval();
>
> #if VM_TRACE
>     tfp->dump(cycles);
> #endif
>     cycles [...]

嗯，如果想看clock变化，这样修改是对的。dump(cycles)指定的是vcd波形里面的时间信息，不影响仿真，只影响波形的输出文件，理论上甚至可以0 1 4 7 9 12 18 22 88 888这样的step都可以

lenged, 2021-07-25:
@poemonsense  @wakafa1 
谢谢解答

closed 2021-07-25 by lenged