OSCPU/NutShell #62: Something wrong when  running on FPGA

zhifanhe, 2021-12-28:
I have downloaded  this project and installed mill. I also run `make` and get the verilog file `build/TopMain.v`.

But when I try to run on FPGA, there is something wrong.
```
cd fpga
make PRJ=myproject BOARD=axu3cg 

# log 
make -C .. BOARD=axu3cg
make[1]: Entering directory '/home/hezf/Documents/chisel_test/nutShell'
make[1]: Nothing to be done for 'verilog'.
make[1]: Leaving directory '/home/hezf/Documents/chisel_test/nutShell'
vivado -nolog -nojournal -notrace -mode batch -source board/axu3cg/mk.tcl -tclargs myproject-axu3cg false

****** Vivado v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source board/axu3cg/mk.tcl -notrace
INFO: [BD_TCL-3] Currently there is no design <nutshell> in project, so creating one...
Wrote  : </home/hezf/Documents/chisel_test/nutShell/fpga/board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.srcs/sources_1/bd/nutshell/nutshell.bd> 
create_bd_design: Time (s): cpu = 00:00:04 ; elapsed = 00:00:07 . Memory (MB): peak = 1254.473 ; gain = 35.852 ; free physical = 8239 ; free virtual = 15436
INFO: [BD_TCL-4] Making design <nutshell> as current_bd_design.
INFO: [BD_TCL-5] Currently the variable <design_name> is equal to "nutshell".
INFO: [BD_TCL-6] Checking if the following IPs exist in the project's IP catalog:  
xilinx.com:ip:axi_protocol_converter:2.1 xilinx.com:ip:system_ila:1.1 xilinx.com:ip:util_vector_logic:2.0  .
INFO: [BD_TCL-6] Checking if the following modules exist in the project's sources:  
NutShell  .
INFO: [IP_Flow 19-5151] The Range '63:0' is present in all ports of the interface 'io_mmio_req_bits'. It is assumed that this is meant to declare an array of interface. If this is not the desired behaviour, switch of this feature by disabling the parameter 'ips.enableInterfaceArrayInference'.
INFO: [IP_Flow 19-5151] The Range '63:0' is present in all ports of the interface 'io_mmio_resp_bits'. It is assumed that this is meant to declare an array of interface. If this is not the desired behaviour, switch of this feature by disabling the parameter 'ips.enableInterfaceArrayInference'.
INFO: [IP_Flow 19-5107] Inferred bus interface 'io_frontend' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
WARNING: [IP_Flow 19-4747] Not inferring AXIMM interface 'io_mem' due to missing master read-channel signals RREADY.
INFO: [IP_Flow 19-5107] Inferred bus interface 'reset' of definition 'xilinx.com:signal:reset:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'clock' of definition 'xilinx.com:signal:clock:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-4728] Bus Interface 'clock': Added interface parameter 'ASSOCIATED_BUSIF' with value 'io_frontend'.
INFO: [IP_Flow 19-4728] Bus Interface 'clock': Added interface parameter 'ASSOCIATED_RESET' with value 'reset'.
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [Device 21-403] Loading part xczu3cg-sfvc784-1-e
create_bd_cell: Time (s): cpu = 00:00:15 ; elapsed = 00:00:24 . Memory (MB): peak = 2257.492 ; gain = 530.430 ; free physical = 7419 ; free virtual = 14763
WARNING: [BD 5-232] No interface pins matched 'get_bd_intf_pins NutShell_0/io_mem'
ERROR: [BD 5-106] Arguments to the connect_bd_intf_net command cannot be empty.
ERROR: [Common 17-39] 'connect_bd_intf_net' failed due to earlier errors.

    while executing
"connect_bd_intf_net -intf_net NutShell_0_io_mem [get_bd_intf_pins NutShell_0/io_mem] [get_bd_intf_pins axi_interconnect_2/S00_AXI]"
    (procedure "create_root_design" line 91)
    invoked from within
"create_root_design """
    (file "board/axu3cg/../../NutShell.tcl" line 400)

    while executing
"source ${tcl_file}"
    (procedure "add_bd" line 2)
    invoked from within
"add_bd ${fpga_dir}/NutShell.tcl"
    invoked from within
"if {${standalone} == "true"} {
  add_bd ${bd_dir}/standalone.tcl
  make_wrapper -files [get_files *system_top.bd] -top
  add_files -norecurse -fileset..."
    (file "board/axu3cg/../common.tcl" line 55)

    while executing
"source ${script_dir}/../common.tcl"
    (file "board/axu3cg/mk.tcl" line 21)
INFO: [Common 17-206] Exiting Vivado at Mon Dec 27 22:26:38 2021...
Makefile:19: recipe for target 'board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.xpr' failed
make: *** [board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.xpr] Error 1
```

Although come across this error , I find .xpr in `fpga/board/axu3cg/build/myproject-axu3cg/`, But when open this project with vivado, nutshell.bd is empty. What is next step? 

Is there any more details about running on fpga? Thanks very much.

wakafa1, 2021-12-28:
I just reproduced your operation from the beginning but there's no error reported. So I suspect it is caused by the mismatched version of Vivado. Could you please try Vivado v2019.1?

zhifanhe, 2021-12-28:
It is so strange that this problem is the same when after changing Vivado v2019.1.

```
make -C .. BOARD=axu3cg
make[1]: Entering directory '/home/hezf/Documents/chisel_test/nutShell'
make[1]: Nothing to be done for 'verilog'.
make[1]: Leaving directory '/home/hezf/Documents/chisel_test/nutShell'
vivado -nolog -nojournal -notrace -mode batch -source board/axu3cg/mk.tcl -tclargs myproject-axu3cg false

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source board/axu3cg/mk.tcl -notrace
INFO: [BD_TCL-3] Currently there is no design <nutshell> in project, so creating one...
Wrote  : </home/hezf/Documents/chisel_test/nutShell/fpga/board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.srcs/sources_1/bd/nutshell/nutshell.bd> 
create_bd_design: Time (s): cpu = 00:00:03 ; elapsed = 00:00:06 . Memory (MB): peak = 1456.859 ; gain = 18.184 ; free physical = 8023 ; free virtual = 15512
INFO: [BD_TCL-4] Making design <nutshell> as current_bd_design.
INFO: [BD_TCL-5] Currently the variable <design_name> is equal to "nutshell".
INFO: [BD_TCL-6] Checking if the following IPs exist in the project's IP catalog:  
xilinx.com:ip:axi_protocol_converter:2.1 xilinx.com:ip:system_ila:1.1 xilinx.com:ip:util_vector_logic:2.0  .
INFO: [BD_TCL-6] Checking if the following modules exist in the project's sources:  
NutShell  .
INFO: [IP_Flow 19-5151] The Range '63:0' is present in all ports of the interface 'io_mmio_req_bits'. It is assumed that this is meant to declare an array of interface. If this is not the desired behaviour, switch of this feature by disabling the parameter 'ips.enableInterfaceArrayInference'.
INFO: [IP_Flow 19-5151] The Range '63:0' is present in all ports of the interface 'io_mmio_resp_bits'. It is assumed that this is meant to declare an array of interface. If this is not the desired behaviour, switch of this feature by disabling the parameter 'ips.enableInterfaceArrayInference'.
INFO: [IP_Flow 19-5107] Inferred bus interface 'io_frontend' of definition 'xilinx.com:interface:aximm:1.0' (from Xilinx Repository).
WARNING: [IP_Flow 19-4747] Not inferring AXIMM interface 'io_mem' due to missing master read-channel signals RREADY.
INFO: [IP_Flow 19-5107] Inferred bus interface 'reset' of definition 'xilinx.com:signal:reset:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-5107] Inferred bus interface 'clock' of definition 'xilinx.com:signal:clock:1.0' (from Xilinx Repository).
INFO: [IP_Flow 19-4728] Bus Interface 'clock': Added interface parameter 'ASSOCIATED_BUSIF' with value 'io_frontend'.
INFO: [IP_Flow 19-4728] Bus Interface 'clock': Added interface parameter 'ASSOCIATED_RESET' with value 'reset'.
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
WARNING: [BD 5-232] No interface pins matched 'get_bd_intf_pins NutShell_0/io_mem'
ERROR: [BD 5-106] Arguments to the connect_bd_intf_net command cannot be empty.
ERROR: [Common 17-39] 'connect_bd_intf_net' failed due to earlier errors.

    while executing
"connect_bd_intf_net -intf_net NutShell_0_io_mem [get_bd_intf_pins NutShell_0/io_mem] [get_bd_intf_pins axi_interconnect_2/S00_AXI]"
    (procedure "create_root_design" line 91)
    invoked from within
"create_root_design """
    (file "board/axu3cg/../../NutShell.tcl" line 400)

    while executing
"source ${tcl_file}"
    (procedure "add_bd" line 2)
    invoked from within
"add_bd ${fpga_dir}/NutShell.tcl"
    invoked from within
"if {${standalone} == "true"} {
  add_bd ${bd_dir}/standalone.tcl
  make_wrapper -files [get_files *system_top.bd] -top
  add_files -norecurse -fileset..."
    (file "board/axu3cg/../common.tcl" line 55)

    while executing
"source ${script_dir}/../common.tcl"
    (file "board/axu3cg/mk.tcl" line 21)
INFO: [Common 17-206] Exiting Vivado at Tue Dec 28 14:11:15 2021...
Makefile:19: recipe for target 'board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.xpr' failed
make: *** [board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.xpr] Error 1

```

And I also change the board pynq. These are some errors. Is there something wrong with my environment setting?

```
hezf@thuos:~/Documents/chisel_test/NutShell/fpga$ make PRJ=myproject BOARD=pynq STANDLONE=true
make -C .. BOARD=pynq
make[1]: Entering directory '/home/hezf/Documents/chisel_test/NutShell'
make[1]: Nothing to be done for 'verilog'.
make[1]: Leaving directory '/home/hezf/Documents/chisel_test/NutShell'
vivado -nolog -nojournal -notrace -mode batch -source board/pynq/mk.tcl -tclargs myproject-pynq false

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source board/pynq/mk.tcl -notrace
ERROR: [Board 49-71] The board_part definition was not found for tul.com.tw:pynq-z2:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7z020clg400-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.

    while executing
"source ${script_dir}/../common.tcl"
    (file "board/pynq/mk.tcl" line 17)
INFO: [Common 17-206] Exiting Vivado at Tue Dec 28 14:21:36 2021...
Makefile:19: recipe for target 'board/pynq/build/myproject-pynq/myproject-pynq.xpr' failed
make: *** [board/pynq/build/myproject-pynq/myproject-pynq.xpr] Error 1

```

wakafa1, 2021-12-29:
In Nutshell, the Vivado project is generated through tcl scripts, so such errors are attributed to Vivado itself convincingly. For example, lack of corresponding board support, lack of IP core support, etc. It is recommended to search Google for help and debug the corresponding tcl scripts

zhifanhe, 2021-12-30:
OK, thanks a lot. Could you show the correct makefile log of axu3cg?

zhifanhe, 2021-12-30:
Done! It seems that the verilog I get is for simulation. To generate fpga verilog 
```
make verilog BOARD=axu3cg
```
Then `cd fpga`, `make PRJ=myproject BOARD=axu3cg` ,and get full vivado project.

wakafa1, 2021-12-30:
Nice to hear that

closed 2021-12-31 by AugustusWillisWang