OSCPU/NutShell #66: Bootgen error for pynq board

zhifanhe, 2021-12-31:
I want to BOOT.BIN for pynq-z2 and follow the step in `/fpga/boot/standalone.md`
Generate verilog for pynq first and create a project in Vivado
```
make verilog BOARD=pynq
cd $(NUTSHELL_HOME)/fpga
make PRJ=myprj BOARD=pynq STANDALONE=true vivado
```
Bitstream and hdf file has been generated successfully.
But after linking ps.hdf and fpga.bit, there is something wrong when `make bootgen PRJ=myprj BOARD=pynq STANDALONE=true`

```
hezf@thuos:~/Documents/chisel_test/NutShell/fpga$ make bootgen PRJ=m3 BOARD=pynq STANDALONE=true
cd boot && hsi -nolog -nojournal -notrace -source mk.tcl -tclargs m3-pynq true

****** hsi v2019.1.3 (64-bit)
  **** SW Build 2644227 on Wed Sep  4 09:44:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

start_gui
invalid command name "common::start_gui"
source mk.tcl -notrace
CRITICAL WARNING: [Board 49-71] The board_part definition was not found for tul.com.tw:pynq-z2:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7z020clg400-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds
open_hw_design: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 864.000 ; gain = 194.016 ; free physical = 10376 ; free virtual = 14046
invalid command name "hsi::utils::get_rp_rm_for_drv"
ERROR: [Hsi 55-1545] Problem running tcl command ::sw_device_tree::generate : invalid command name "hsi::utils::get_rp_rm_for_drv"
    while executing
"hsi::utils::get_rp_rm_for_drv $drv_handle"
    (procedure "set_drv_def_dts" line 14)
    invoked from within
"set_drv_def_dts $drv_handle"
    (procedure "gen_peripheral_nodes" line 54)
    invoked from within
"gen_peripheral_nodes $drv_handle "create_node_only""
    ("foreach" body line 3)
    invoked from within
"foreach drv_handle [get_drivers] {
        # generate the default properties
        gen_peripheral_nodes $drv_handle "create_node_only"
        gen_r..."
    (procedure "::sw_device_tree::generate" line 3)
    invoked from within
"::sw_device_tree::generate device_tree"
ERROR: [Hsi 55-1442] Error(s) while running TCL procedure generate()
ERROR: [Common 17-39] 'generate_target' failed due to earlier errors.

    while executing
"generate_target -dir ${build_dir}/dts"
    (file "mk.tcl" line 77)
hsi% quit
INFO: [Common 17-206] Exiting hsi at Fri Dec 31 22:00:56 2021...
```
Why ‘genrate_target’ failed? The pynq-z2 is also in board_files, why  'CRITICAL WARNING'? 
Thanks a lot and happy new year!

wakafa1, 2021-12-31:
We mistakenly introduced an absolute path that shouldn't appear in NutShell/fpga/boot/mk.tcl. You can try to clone the https://github.com/xilinx/device-tree-xlnx project, checkout to hash 73e546e, and then modify the device_tree_repo_path in mk.tcl:24 to your path

zhifanhe, 2022-01-01:
Thanks a lot. I clone the project 'device-tree-xlnx', but can't find the hash 73e546e. After change the `device_tree_repo_path`, the problem is the same, `generate device_tree` failed.

zhifanhe, 2022-01-01:
Done! It is really a version error.  The default `device-tree-xlnx` is 2020.2. I down the [19.1 version](https://github.com/Xilinx/device-tree-xlnx/releases/tag/xilinx-v2019.1), and change line 24 in mk.tcl. 
It seems a new year gift!

closed 2022-01-03 by wakafa1