OSCPU/NutShell #68: Can not boot RISC-V subsystem on axu3eg Board ?

zhifanhe, 2022-01-04:
I sucessfully build system_top.bit file for axu3eg Board and use petalinux to boot linux in arm core. 
Then I follow the step in README.md-Boot NutShell.
```
gcc -O2 -o ddr-loader ddr-loader.c
./ddr-loader axu3cg linux.bin # in ready-to-run folder
```
The output shows:
```
Welcome to minicom 2.7.1

OPTIONS: I18n
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyUL1, 08:17:02

Press CTRL-A Z for help on special keys

bbl loader
freq-mhz = 500
CLINT: set frequency to 500 MHz
```

if use `./ddr-loader zcu102 linux.bin`,  it shows 'Segmentation fault (core dumped)'.
```
xilinx@pynq:/home/nutshell$ sudo ./ddr-loader zcu102 linux.bin
board = zcu102, ddr_size = 0x80000000, ddr_base = 0x800000000
mapping paddr 0x800000000 to vaddr 0x7f3d748000
mapping paddr 0x[  231.709977] Bad mode in Error handler detected on CPU2, code 0xbf000000 -- SError
[  231.722988] Internal error: Oops - bad mode: 0 [#1] SMP
[  231.728202] Modules linked in: icfc(O) uio_pdrv_genirq
[  231.733336] CPU: 2 PID: 3480 Comm: ddr-loader Tainted: G           O    4.14.0-xilinx-v2018.3 #1
[  231.742107] Hardware name: xlnx,zynqmp (DT)
[  231.746275] task: ffffffc87b15a600 task.stack: ffffff800e560000
[  231.752177] PC is at 0x7fbd811f90
[  231.755474] LR is at 0x55692e9da8
[  231.758773] pc : [<0000007fbd811f90>] lr : [<00000055692e9da8>] pstate: 60000000
70010[0 0 02 3t1o. 7v6a6d1d5r0 ]0 xs7pf b:d 80e0c0000000
 fldoaa76ac0
2ing linux.bin [t o  2o3f1f.s7e7t2 4=0 00]x 0x
 9: 0000007fda276ac0 x28: 0000000000000000
[  231.780472] x27: 0000000000000000 x26: 0000000000000000
[  231.785768] x25: 0000007fda277e03 x24: 00000055692fac30
[  231.791071] x23: 0000000000000000 x22: 0000007fda276c58
[  231.796374] x21: 00000055692fa000 x20: 00000055692fac90
[  231.801678] x19: 00000055692fb028 x18: 0000007fda276977
[  231.806982] x17: 0000007fbd811f90 x16: 00000055692faf88
[  231.812285] x15: 0000000000000010 x14: 0000000000000000
[  231.817588] x13: 0000000000000000 x12: 0000000000000020
[  231.822883] x11: 0000007fda276978 x10: 0000000000000000
[  231.828178] x9 : 00000000ffffffff x8 : 0000000000000040
[  231.833473] x7 : 000000000000000a x6 : 000000000000000a
[  231.838768] x5 : 000000557cc58282 x4 : 0000000000000022
[  231.844063] x3 : 0000000000000000 x2 : 7f70de81dbc25800
[  231.849358] x1 : 0000007fbd8ec000 x0 : 00000000000003e8
[  231.854662] Process ddr-loader (pid: 3480, stack limit = 0xffffff800e560000)
[  231.861694] ---[ end trace 2843468e7585934a ]---
Segmentation fault (core dumped)
```

wakafa1, 2022-01-05:
The error message shows that it is a violation of memory access. Please debug the triggered error address in ddr-loader.c, and confirm whether the address is a hole according to the board information.

wakafa1, 2022-01-05:
By the way, our tested board is axu3cg instead of axu3eg. There may exist some differences.

zhifanhe, 2022-01-06:
OK, thanks a lot.

closed 2022-01-25 by wakafa1