OSCPU/NutShell #69: 如何生成适合于自定义板卡的RV_BOOT.bin文件

hiterinbj, 2022-01-06:
在使用nutshell加载risc-v子系统时，需要加载bin文件，用了ready-to-run里面的linux.bin,在minicom中只显示了bbl loader何frequency，换了香山里面的ready-to-run 里面的linux.bin，能打印hello。RISC-V World! 之后又打印了一堆寄存器地址，最后停留在kernel panic not syncing：Attempted to kill init exitcode=0x00000004。
在查阅之前的issues，看到您之前回复其他人 “你好，这边建议使用我们预编好的内核镜像，在 NutShell/fpga/boot/pynq/RV_BOOT.bin；如果想要自行生成，Linux 配置请使用 riscv-linux/arch/riscv/configs/fpga_defconfig，其中一些选项需要根据实际情况酌情修改；riscv-pk 方面，请在 riscv-pk/dts 中使用合适的设备树，然后 make 即可，最终在 riscv-pk/build 目录下会得到 bbl.bin 文件，即为我们想要的 RV_BOOT.bin（名字默认的意思是后续的流程会直接使用“RV_BOOT.bin”文件名，因此需要手动将 bbl.bin 重命名为 RV_BOOT.bin）”
在riscv-linux/arch/riscv/configs/中没有fpga_defconfig 只有defconfig，在riscv-pk repo中也没有dts，在生成bbl.bin中设备树怎么用，有没有资料参考一下 ，用于生成自定义fpga板卡的RV_BOOT.bin

wakafa1, 2022-01-06:
1. 当打印出 “hello RISC-V World!“ 后，说明该 linux.bin 已经成功运行，后续出现 panic 是因为 Linux 不允许 init 进程自动退出，但我们的 init 进程是自己写的一个简单 hello.c 程序它运行完后会退出，因此可以无视该 panic
2. 想要生成 RV_BOOT.bin，需要 riscv-pk(bbl)、riscv-linux 以及可选的 riscv-rootfs，其中 riscv-pk 的 dts 文件夹下有一些示例的设备数，需要您根据自己的板卡 block design 信息自行修改配置；riscv-linux 您根据板卡信息切换到不同的分支，可以看到有 fpga_defconfig，其中内容也需要根据板卡情况酌情修改

hiterinbj, 2022-01-06:
好的 谢谢 我先看看~

hiterinbj, 2022-01-07:
虽然可以无视panic，在minicom中显示hello。说明linux.bin已经成功运行，看您操作riscv子系统起来之后，通过ssh的指令远程过去的，我这边ssh不到192.168.1.2，是和我用的香山的linux.bin有关吗？那我该如何进入riscv子系统呢？烦请告知 谢谢

wakafa1, 2022-01-07:
演示的 gif 是一个比较复杂的启动过程，首先启动 Linux Kernel 然后进入 Debian，另外还在 FPGA 上配置了网络环境，因此才能使用网络连接；在你这边的情况下，可以使用 Busybox 作为 /init 来启动一个交互式的 Shell；Busybox 的内容参见 riscv-rootfs

wakafa1, 2022-01-07:
关于构建 Linux Kernel，这里有一份更详细的文档：
https://github.com/OpenXiangShan/XiangShan-doc/blob/main/tutorial/others/Linux%20Kernel%20%E7%9A%84%E6%9E%84%E5%BB%BA.md

hiterinbj, 2022-01-07:
谢谢~

closed 2022-01-25 by wakafa1