OSCPU/NutShell #71: ram.cpp模型中的ram，当前周期读出rdata，不太现实吧？

trueif, 2022-01-25:
在ram.cpp中，用ram_read_helper读出了ram的数据，然后当周期就返回给ram.v的输出rdata。这个有点不太现实了吧。如果我改写成非阻塞赋值，写在always块中，延迟一个周期回数据给rdata，跑出来的结果就不对了：

` 31 module RAMHelper(
 32   input         clk,
 33   input         en,
 34   input  [63:0] rIdx,
 35   output reg [63:0] rdata,
 36   input  [63:0] wIdx,
 37   input  [63:0] wdata,
 38   input  [63:0] wmask,
 39   input         reset,
 40   input         wen
 41 );
 42   reg [63:0] rdata_this;
 43   assign rdata_this = ram_read_helper(en, rIdx);
 44   always @(posedge clk) begin
 45     ram_write_helper(wIdx, wdata, wmask, wen && en);
 46     rdata <= rdata_this;
 47   end
 48
 49 endmodule
`

请问本设计上FPGA板或者要进行ASIC设计时，ram改如何处理？是不是有参数可以调节ram的实际延迟周期数？

wakafa1, 2022-01-25:
仿真下使用了 0 延迟，实际 FPGA/ASIC 会设置“FPGAPlatform” 参数，直接把访存信号转换为 AXI4 总线请求接到顶层 IO；
至于您修改出错的问题，请注意核内的信号处理：访存是握手信号，如果只是数据晚一拍但是 valid 信号不变，肯定是不对的，请注意此处代码https://github.com/OSCPU/NutShell/blob/59544e07899cdfa22fbb07e1a9f92069e734f5f0/src/main/scala/device/AXI4Slave.scala#L74

poemonsense, 2022-01-25:
可以试试打开dramsim的选项，然后在ram.cpp里面改成一个固定延迟

trueif, 2022-01-25:
谢谢两位。我后来发现是我的RAM没写好，我可以assign rdata = rdata_this[rIdx[63:0]]; 即可，这样做出来的是当前周期即可读取的RAM。跑microbench已经通过。

现在在搞linux.bin。目前发现verilator跑这个bin会崩溃。可能需要换个仿真器。。。

closed 2022-01-25 by trueif

poemonsense, 2022-01-25:
大概率是自己代码的问题，不是verilator的问题

trueif, 2022-01-27:
> 

你说的没错。确实是我的ram.v没写好。现在改好了。谢谢啊。

trueif, 2022-01-27:
> 

“现在在搞linux.bin。目前发现verilator跑这个bin会崩溃。可能需要换个仿真器。。。”

是用ram.cpp读取img文件的话，verilator不会crash。但是我把img固化写在ram.v的话，verilator就会crash。如果我把img的内容减半固化在ram.v，则verilator不crash。

trueif, 2022-01-27:
> 大概率是自己代码的问题，不是verilator的问题

“现在在搞linux.bin。目前发现verilator跑这个bin会崩溃。可能需要换个仿真器。。。”

是用ram.cpp读取img文件的话，verilator不会crash。但是我把img固化写在ram.v的话，verilator就会crash。如果我把img的内容减半固化在ram.v，则verilator不crash。