OSCPU/NutShell #73: OOO make failed

cnjsdfcy, 2022-02-10:


cnjsdfcy, 2022-02-10:
Hi,

I changed CORE in Makefile from **_inorder_** to _**ooo**_, and got a compile error as below:

mkdir -p build
mill chiselModule.runMain top.TopMain -td build --output-file TopMain.v --infer-rw NutShellFPGATop --repl-seq-mem -c:NutShellFPGATop:-o:build/TopMain.v.conf BOARD=sim   CORE=ooo  
[63/63] chiselModule.runMain 
====== Settings = (sim, ooo) ======
EnableDebug = true
EnableILA = true
EnableMultiIssue = true
EnableOutOfOrderExec = true
EnableRVC = true
FPGAPlatform = false
HasDTLB = true
HasDcache = true
HasITLB = true
HasIcache = true
HasL2cache = true
HasPrefetch = true
IsRV32 = false
MMIOBase = 0x40000000
MMIOSize = 0x40000000
MemMapBase = 0x0
MemMapRegionBits = 0
MmodeOnly = false
NrExtIntr = 1
ResetVector = 0x80000000
Exception in thread "main" chisel3.internal.ChiselException: Connection between left (DecoupledIO(IO io_in_1 in Arbiter)) and source (DecoupledIO(IO in unelaborated Cache)) failed @.bits.id: Right Record missing field (id).
        at ... ()
        at nutcore.Cache.<init>(Cache.scala:496)
        at nutcore.Cache$.$anonfun$apply$5(Cache.scala:670)
        at chisel3.Module$.do_apply(Module.scala:52)
        at nutcore.Cache$.apply(Cache.scala:670)
        at nutcore.NutCore.$anonfun$new$11(NutCore.scala:129)
        at chisel3.Data.$anonfun$$less$greater$1(Data.scala:644)
        at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
        at chisel3.internal.prefix$.apply(prefix.scala:32)
        at chisel3.Data.$less$greater(Data.scala:644)
        at nutcore.NutCore.<init>(NutCore.scala:128)
        at system.NutShell.$anonfun$nutcore$1(NutShell.scala:53)
        at chisel3.Module$.do_apply(Module.scala:52)
        at system.NutShell.<init>(NutShell.scala:53)
        at sim.SimTop.$anonfun$soc$1(NutShellSim.scala:40)
        at chisel3.Module$.do_apply(Module.scala:52)
        at sim.SimTop.<init>(NutShellSim.scala:40)
        at top.TopMain$.$anonfun$new$6(TopMain.scala:68)
        at ... ()
        at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
1 targets failed
chiselModule.runMain subprocess failed
Makefile:23: recipe for target 'build/TopMain.v' failed
make: *** [build/TopMain.v] Error 1

AugustusWillisWang, 2022-03-02:
Previous change broke ooo config, #74 should have fixed it.

cnjsdfcy, 2022-03-08:
Thanks, I fixed it by myself locally some days ago.
Also, I found and fixed a embedded building error, pls check below:
```
diff --git a/src/main/scala/nutcore/mem/EmbeddedTLB.scala b/src/main/scala/nutcore/mem/EmbeddedTLB.scala
index a6303e8..a2e58cd 100644
--- a/src/main/scala/nutcore/mem/EmbeddedTLB.scala
+++ b/src/main/scala/nutcore/mem/EmbeddedTLB.scala
@@ -407,6 +407,7 @@ class EmbeddedTLBEmpty(implicit val tlbConfig: TLBConfig) extends TlbModule {
 }
 
 class EmbeddedTLB_fake(implicit val tlbConfig: TLBConfig) extends TlbModule with HasTLBIO {
+  io.mem := DontCare
   io.out <> io.in
   io.csrMMU.loadPF := false.B
   io.csrMMU.storePF := false.B
@@ -427,9 +428,10 @@ object EmbeddedTLB {
     } else {
       val tlb = Module(new EmbeddedTLB_fake)
       tlb.io.in <> in
+      tlb.io.mem <> mem
       tlb.io.flush := flush
       tlb.io.csrMMU <> csrMMU
-      mem := DontCare
+      //mem := DontCare
       tlb
     }
   }
```

AugustusWillisWang, 2022-03-14:
LGTM

closed 2022-03-26 by cnjsdfcy