OSCPU/NutShell #79: Unable to compile the project if TLBs are disabled

Donlon, 2022-03-21:
When I set `HasDTLB` and `HasITLB` to `false` in [Settings.scala](https://github.com/OSCPU/NutShell/blob/cf7963435b824106690152507bb9312c159b8d5e/src/main/scala/top/Settings.scala#L48), the project could not be compiled. The output of `make` command is

```
NutShell $ make PRJ=myproject BOARD=pynq STANDALONE=true
mkdir -p build
mill chiselModule.runMain top.TopMain -td build --output-file TopMain.v --infer-rw NutShellFPGATop --repl-seq-mem -c:NutShellFPGATop:-o:build/TopMain.v.conf BOARD=pynq CORE=inorder
[63/74] chiselModule.compile
[info] compiling 77 Scala sources to /home/lighthouse/rv/NutShell/out/chiselModule/compile.dest/classes ...
[warn] 2105 feature warnings; re-run with -feature for details
[warn] one warning found
[info] done compiling
[74/74] chiselModule.runMain
====== Settings = (pynq, inorder) ======
EnableDebug = true
EnableILA = true
EnableMultiIssue = false
EnableOutOfOrderExec = false
EnableRVC = true
FPGAPlatform = true
HasDTLB = false
HasDcache = true
HasITLB = false
HasIcache = true
HasL2cache = true
HasPrefetch = true
IsRV32 = false
MMIOBase = 0xe0000000
MMIOSize = 0x20000000
MemMapBase = 0x10000000
MemMapRegionBits = 28
MmodeOnly = true
NrExtIntr = 3
ResetVector = 0x60000000
[deprecated] class utils.FlushableQueueIO (1 calls): Unable to automatically infer cloneType on class utils.FlushableQueueIO. cloneType is now implemented by the Chisel compiler plugin so please ensure you are using it in your build. If you cannot use the compiler plugin or you are using it and you still see this message, please file an issue and let us know. For those not using the plugin, here is the 'runtime reflection' cloneType error message: constructor has parameters (entries) that are not both immutable and accessible. Either make all parameters immutable and accessible (vals) so cloneType can be inferred, or define a custom cloneType method.
[warn] There were 1 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Exception in thread "main" firrtl.passes.PassExceptions:
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.bits.size <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.bits.wdata <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.bits.cmd <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.resp.ready <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.bits.addr <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.bits.wmask <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake]  Reference io is not fully initialized.
   : io.mem.req.valid <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.bits.size <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.bits.wdata <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.bits.cmd <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.resp.ready <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.bits.addr <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.bits.wmask <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module EmbeddedTLB_fake_1]  Reference io is not fully initialized.
   : io.mem.req.valid <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake_1 is not fully initialized.
   : EmbeddedTLB_fake_1.io.mem.resp.bits.cmd <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake is not fully initialized.
   : EmbeddedTLB_fake.io.mem.req.ready <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake is not fully initialized.
   : EmbeddedTLB_fake.io.mem.resp.valid <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake_1 is not fully initialized.
   : EmbeddedTLB_fake_1.io.mem.resp.valid <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake_1 is not fully initialized.
   : EmbeddedTLB_fake_1.io.mem.resp.bits.rdata <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake is not fully initialized.
   : EmbeddedTLB_fake.io.mem.resp.bits.rdata <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake is not fully initialized.
   : EmbeddedTLB_fake.io.mem.resp.bits.cmd <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[EmbeddedTLB.scala 428:23] : [module NutCore]  Reference EmbeddedTLB_fake_1 is not fully initialized.
   : EmbeddedTLB_fake_1.io.mem.req.ready <= VOID
firrtl.passes.PassException: 22 errors detected!
1 targets failed
chiselModule.runMain subprocess failed
make: *** [Makefile:24: build/TopMain.v] Error 1
```

Running `make` with `CORE=embedded` (`make PRJ=myproject BOARD=pynq CORE=embedded STANDALONE=true`) produces the same error. Could you give some help on it?

AugustusWillisWang, 2022-03-25:
Please try https://github.com/OSCPU/NutShell/issues/73

Donlon, 2022-03-25:
This patch works for me. Thanks

closed 2022-03-25 by Donlon