OSCPU/NutShell #80: Run on FPGA  在使用vivado generate bitstream 时报错

dujiaming1107, 2022-03-23:
vivado 版本是2019.1
执行
```
cd fpga
make PRJ=myproject BOARD=pynq STANDALONE=true
```
成功构建vivado，但是在打开工程generate bitstream时出现下面的错误
```
SynthesisOut-of-Context Module Runsnutshellnutshell_NutShell_0_0_synth_1[Synth 8-439] module 'array_ext' not found ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":24646]
[Synth 8-6156] failed synthesizing module 'array' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":24632]
[Synth 8-6156] failed synthesizing module 'SRAMTemplate' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":16]
[Synth 8-6156] failed synthesizing module 'BPU_inorder' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":146]
[Synth 8-6156] failed synthesizing module 'IFU_inorder' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":424]
[Synth 8-6156] failed synthesizing module 'Frontend_inorder' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":2453]
[Synth 8-6156] failed synthesizing module 'NutCore' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":14128]
[Synth 8-6156] failed synthesizing module 'NutShell' ["C:/home/dujiaming/Desktop/xs-env/NutShell/build/TopMain.v":22014]
[Synth 8-6156] failed synthesizing module 'nutshell_NutShell_0_0' ["C:/home/dujiaming/Desktop/xs-env/NutShell/fpga/board/axu3cg/build/myproject-axu3cg/myproject-axu3cg.srcs/sources_1/bd/nutshell/ip/nutshell_NutShell_0_0/synth/nutshell_NutShell_0_0.v":58]
[Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
```
尝试把BOARD改为axu3cg，出现同样的错误。
请问这个错误出现的原因可能是哪一部分的问题？

Donlon, 2022-03-24:
同样遇到了这个问题，不过我发现这个模块和几个叫 array_n_ext 的模块都是用一个专门的脚本[生成](https://github.com/OSCPU/NutShell/commit/a7504a53ac47aefee932504ac884e0c27c3fc736)的，但后来不知道为什么在[cf79634](https://github.com/OSCPU/NutShell/commit/cf7963435b824106690152507bb9312c159b8d5e)里被移除掉了，你可以看下之前的版本。

AugustusWillisWang, 2022-03-25:
我们更新 Nutshell 使用的 Chisel 版本之后没有重新测试 FPGA 脚本。您可以先用 https://github.com/OSCPU/NutShell/releases/tag/release-211228 这个稳定分支看看？

closed 2022-11-23 by du-jiaming

qimingchu, 2023-01-10:
> 我们更新 Nutshell 使用的 Chisel 版本之后没有重新测试 FPGA 脚本。您可以先用 https://github.com/OSCPU/NutShell/releases/tag/release-211228 这个稳定分支看看？

稳定分支在使用vivado synthesis的时候会出现报错[Unable to synthesize the design in Vivado 2021.2 #78]
生成verilog的命令是``` make PRJ=myproject BOARD=pynq STANDALONE=true```，其余都是默认配置，我用的是vivado版本是22.2，出现了同样的错误
[image]
不知道现在有办法能解决吗

qimingchu, 2023-01-16:
> 稳定分支在使用vivado synthesis的时候会出现报错[Unable to synthesize the design in Vivado 2021.2 #78] 生成verilog的命令是` make PRJ=myproject BOARD=pynq STANDALONE=true`，其余都是默认配置，我用的是vivado版本是22.2，出现了同样的错误 [image] [...]

解决了，在`EmbeddedTLBMD`模块中的数组`tlbmd_0` `tlbmd_2` `tlbmd_3`前都添加`(* ram_style = "registers" *) `就能完成综合
下面是个例子，如
```
(* ram_style = "registers" *) 
  reg [120:0] tlbmd_0 [0:0];
```
[方法来源](https://github.com/OSCPU/NutShell/issues/78)