OSCPU/NutShell #81: New transient execution attack on NutShell.

JaewonHur, 2022-03-25:
Hi,
Currently, I'm running a tool to find transient execution vulnerabilities in an open-source CPU.
While running it on NutShell, I found a new sort of transient execution vulnerability in NutShell.

While transient execution vulnerabilities are not considered a lot (and fixed) in today's open-source CPUs,
At least, I think it would be meaningful for the CPU developers, and software developers on top of that to know which transient execution attacks are possible on their CPU. 

The attached PoC is a Spectre type attack. To be specific, the attack is based on two points: **1) port contention on division unit**, and **2) variable latency of division operation depending on input value**.  When a victim contains an instruction sequence as follow, 
```
... /* delayed computation which outputs a1 (e.g., memory, multiply) */
(1) div a0, a0, a1       /* instruction using division unit */
(2) beq a0, a2, 1f       /* misspeculated branch depending on a0 */
    lw t0, [secret]
(3) remuw t0, t0, a3     /* instruction using division unit */
...
```
If the dependency of instruction (3) is resolved earlier than instruction (1),  it can be dispatched earlier than (1), even if it is a transient instruction. Then, (1) can only be dispatched after the computation of (3) completes. Since the misspeculated branch (2) depends on (1), the branch is resolved only when (1) finishes. However, since the completion time of (3) depends on the secret, the branch resolution time finally depends on the secret. Thus, the attacker can reconstruct the secret by measuring the branch resolution time.

I attach the PoC file below, and the attack can be reproduced as follow:
```unzip poc.zip
cd poc
cp nutshell.patch $NUTSHELL_HOME/
cp nemu.patch $NEMU_HOME/
cp nexus-am.patch $AM_HOME/
cp -r birgus $AM_HOME/apps

/* patch */
cd $NUTSHELL_HOME; patch -p1 < nutshell.patch
cd $NEMU_HOME; patch -p1 < nemu.patch
cd $AM_HOME; patch -p1 < nexus-am.patch

/* build poc */
cd $AM_HOME/apps/birgus; make ARCH=riscv64-nutshell

/* build cpu */
cd $NUTSHELL_HOME; make ./build/emu
./build/emu -i $AM_HOME/apps/birgus/build/birgus-riscv64-nutshell.bin -b 0 -e 0 
```

[poc.zip](https://github.com/OSCPU/NutShell/files/8346843/poc.zip)

AugustusWillisWang, 2022-03-25:
NutShell is an in-order core. IMHO, it should not have transient execution vulnerability you mentioned above?

JaewonHur, 2022-03-25:
I found the vulnerability in the out-of-order version of NutShell.
Does it have another name than NutShell?

AugustusWillisWang, 2022-03-25:
That problem does exist in the out-of-order core.

The "out-of-order version of NutShell" is a personal project (called Argo). :)

JaewonHur, 2022-03-25:
Thanks!
Let me report other issues if I find.

closed 2022-03-25 by JaewonHur