OSCPU/NutShell #88: 关于使用nemu仿真RV_BOOT.bin的报错

xddcore, 2022-05-08:
您好，最近在进行nemu仿真运行RV_BOOT.bin是遇到了如下问题:
我参考了https://github.com/OpenXiangShan/NEMU/issues/22的中的问题解决办法，但考虑到nutshell这边fork的nemu和xiangshan的nemu在makefile上有较大改动。所以还是创建了这个issues，感谢您的帮助。

我的操作如下:

```
nemu clone from https://github.com/OSCPU/nemu
export ISA=riscv64
```
然后在/OSCPU/nemu根目录下make -j
最后生成/OSCPU/nemu/build
执行:
` ./build/riscv64-nemu-interpreter ../NutShell/fpga/boot/pynq/RV_BOOT.bin `
得到报错如下:
```
address (0xe000002c) is out of bound {???} [0x00000000, 0x00000000] at pc = 0x000000008000266a
$0: 0x0000000000000000 ra: 0x0000000080001c0c sp: 0x000000008000bc30 gp: 0x0000000000000000 
tp: 0x0000000000000000 t0: 0x0000000000000009 t1: 0x0000000080006c78 t2: 0x0000000000000025 
s0: 0x000000008000bc41 s1: 0x0000000000000000 a0: 0x0000000000000062 a1: 0x0000000000000100 
a2: 0x00000000800071f4 a3: 0x000000008000bd78 a4: 0x00000000e0000000 a5: 0x00000000e0000000 
a6: 0x0000000000000000 a7: 0x000000000000000c s2: 0x0000000000000000 s3: 0x0000000000000000 
s4: 0x0000000000000000 s5: 0x0000000000000000 s6: 0x0000000000000000 s7: 0x0000000000000000 
s8: 0x0000000000000000 s9: 0x0000000000000000 s10: 0x0000000000000000 s11: 0x0000000000000000 
t3: 0x00000000800071f4 t4: 0x000000008000bc40 t5: 0x0000000000000015 t6: 0xffffffffffffffff 
pc: 0x000000008000266a mstatus: 0x0000000000000000 mcause: 0x0000000000000000 mepc: 0x0000000000000000
                       sstatus: 0x0000000000000000 scause: 0x0000000000000000 sepc: 0x0000000000000000
[src/monitor/cpu-exec.c,35,monitor_statistic] total guest instructions = 96815
riscv64-nemu-interpreter: src/device/io/map.c:21: check_bound: Assertion `map != ((void *)0) && addr <= map->high && addr >= map->low' failed.
已放弃 (核心已转储)
```
[image]

wakafa1, 2022-05-09:
NutShell/fpga/boot/pynq/RV_BOOT.bin 是给 PYNQ 开发板用的内核镜像文件。在仿真下，请使用 ready-to-run 目录下的 bin

xddcore, 2022-05-10:
好的，感谢您的慷慨帮助！

closed 2022-05-10 by xddcore