OSCPU/NutShell #89: 在本项目目录ready-to-run下的microbench.bin中的代码段运行指令不明确

AppDoraemon, 2022-05-12:
我们使用了您项目下的microbench.bin结合difftest对我们的riscv处理器进行了测试。在测试过程中发现处理器执行到了pc:0x80004638，这个地址对应的指令是0005006b。我使用了gcc工具链进行反汇编并没有确定这条指令的具体功能，在官方文档中也未能确认到合适的指令。所以想请问您是否知晓这条指令的具体内容及执行逻辑。感谢您的解答。

AppDoraemon, 2022-05-12:
这是反汇编后得到的内容，地址位于80004638的前后指令。
`    4624:	0540006f          	j	0x4678
    4628:	ff010113          	addi	sp,sp,-16
    462c:	00113423          	sd	ra,8(sp)
    4630:	00050593          	mv	a1,a0
    4634:	00050513          	mv	a0,a0
    4638:	0005006b          	0x5006b
    463c:	00002517          	auipc	a0,0x2
    4640:	8a450513          	addi	a0,a0,-1884 # 0x5ee0
    4644:	729000ef          	jal	ra,0x556c
    4648:	0000006f          	j	0x4648
    464c:	ff010113          	addi	sp,sp,-16
    4650:	00113423          	sd	ra,8(sp)
    4654:	014000ef          	jal	ra,0x4668
    4658:	00002517          	auipc	a0,0x2
    465c:	89d50513          	addi	a0,a0,-1891 # 0x5ef5
    4660:	d00ff0ef          	jal	ra,0x3b60
    4664:	fc5ff0ef          	jal	ra,0x4628`

wakafa1, 2022-05-12:
这应当是一条内部假指令 _halt，用于使果壳仿真停机。您可以查看 nexus-am 项目代码，并自行编译 microbench

AppDoraemon, 2022-05-12:
> 这应当是一条内部假指令 _halt，用于使果壳仿真停机。您可以查看 nexus-am 项目代码，并自行编译 microbench

好的，感谢您的解答。