OSCPU/NutShell #92: difftest scala消失了。这个在香山里面有。建议提交代码前clean一下，或者删掉 difftest库。

bigdot123456, 2022-07-02:
difftest：
https://github.com/OpenXiangShan/difftest
香山里面自己有diff代码，果壳没有对应的代码上传。

报错信息
[info] compiling 77 Scala sources to /home/bigdot/git2/NutShell/out/chiselModule                                                                                                        /compile.dest/classes ...
[error] /home/bigdot/git2/NutShell/src/main/scala/device/AXI4UART.scala:24:8: no                                                                                                        t found: object difftest
[error] import difftest._
[error]        ^
[error] /home/bigdot/git2/NutShell/src/main/scala/device/AXI4UART.scala:26:67: n                                                                                                        ot found: type UARTIO
[error] class AXI4UART extends AXI4SlaveModule(new AXI4Lite, _extra = new UARTIO

bigdot123456, 2022-07-02:
删掉difftest后，需要将UARTIO单独形成模块。

bigdot123456, 2022-07-02:
如果用香山的库，可以编译通过，但是会出这种warning：
[deprecated] class utils.FlushableQueueIO (1 calls): Unable to automatically infer cloneType on class utils.FlushableQueueIO. cloneType is now implemented by the Chisel compiler plugin so please ensure you are using it in your build. If you cannot use the compiler plugin or you are using it and you still see this message, please file an issue and let us know. For those not using the plugin, here is the 'runtime reflection' cloneType error message: constructor has parameters (entries) that are not both immutable and accessible. Either make all parameters immutable and accessible (vals) so cloneType can be inferred, or define a custom cloneType method.

TuringKi, 2022-07-03:
你大概是没有更新子模块：git submodule init && git submodule update