OSCPU/NutShell #98: 关于执行bootgen命令生成BOOT.BIN文件出现的错误

L0stMyN4me, 2022-09-09:
你好！

我根据文件[standalone.md](https://github.com/OSCPU/NutShell/blob/master/fpga/boot/standalone.md)进行操作来生成BOOT.BIN，在更改完设备树路径并且根据要求创建了system_top_wrapper.hdf和system_top_wrapper.bit的软连接后
再次执行命令make bootgen PRJ=myprj BOARD=pynq STANDALONE=true 
但是出现如下错误：

cd boot && hsi -nolog -nojournal -notrace -source mk.tcl -tclargs myproject-pynq true

****** hsi v2019.1.3 (64-bit)
  **** SW Build 2644227 on Wed Sep  4 09:44:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

start_gui
invalid command name "common::start_gui"
source mk.tcl -notrace
CRITICAL WARNING: [Board 49-71] The board_part definition was not found for tul.com.tw:pynq-z2:part0:1.0. The project's board_part property was not set, but the project's part property was set to xc7z020clg400-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds
WARNING: not supported pl_clk:
WARNING: not supported pl_clk:
INFO: [Common 17-206] Exiting hsi at Fri Sep  9 03:51:55 2022...

错误提示not supported pl_plk;但是成功生成了BOOT.BIN
请问这是因为fsbl-loader编译时的问题还是其他原因呢？

之后将BOOT.BIN和RV_BOOT.BIN放入SD卡中上板启动串口打印信息如下
[image]
卡在了fsbl-loader.c的main函数中的最后一步

  xil_printf("Reset RISC-V core.\r\n");
	/* release RISC-V reset signal */
	gpio_base[0] = 0x1;

	while(1);

	return XST_SUCCESS;
请问这里复位不成功是什么原因呢？

还请指导一下！
感谢！

wakafa1, 2022-09-13:
使用我们提供的 BOOT.bin，您可以成功启动嘛？

L0stMyN4me, 2022-09-13:
> 使用我们提供的 BOOT.bin，您可以成功启动嘛？

使用你们的BOOT.bin能够成功看到打印信息

wakafa1, 2022-09-13:
我刚刚复现了一下，发现运行日志与您分享的是一样的，这些 warning 不需要关心。
看现象像是 PL 没有被成功复位，也有可能是 PL 上面的 bit 有问题，这个可能比较难调试

L0stMyN4me, 2022-09-15:
> 我刚刚复现了一下，发现运行日志与您分享的是一样的，这些 warning 不需要关心。 看现象像是 PL 没有被成功复位，也有可能是 PL 上面的 bit 有问题，这个可能比较难调试

感谢您的回复，接下来我们慢慢排查