OSCPU/NutShell #114: 关于地址长度的一些疑问

SeddonShen, 2023-04-09:
在学习NutShell源代码时，对地址长度这块有点疑问，想请教一下：
在默认参数的情况下，也就是64位的顺序核，在不开启虚拟地址的情况下，处理器中很多地方都已经默认以VAddrBits=39来作为地址了，但是在不打开虚拟地址的情况下，是不是该按照64位寻址呢？
这是我构造的一个非对齐访存的测例，使用Difftest联合NEMU进行差分测试，出现了与Difftest不一致的情况：
## ⾮对⻬访存写mtval的问题
```
li t0, 0x80 0000 00A5
ld x15,119(t0)
```
上述问题表现为，NutShell中对异常处理的地址进⾏了截断，⾼位（39位以上）将不存在，从⽽导致mtval出现问题。 该问题已经在参考模型验证和Difftest中尝试复现成功，具体表现为mtval的值应为：
0x000000800000011c
⽽NutShell表现值为：
0x000000000000011c
请问设计时默认按照39位是否符合RISC-V的规范呢，因为对这一块不太熟序，所以不知道是不是设计时的特殊考虑？

Lemover, 2023-04-09:
ISA 中会定义 XLEN 等常量，而在处理器中，会存在 VAddrBits 和 PAddrBits 两个常量，分别对应虚拟地址空间的地址长度（Sv39等），和真实物理地址空间的长度（比如最大只支持 4GB）。

严格来说，mtval 需要完整的 64 位数据，但在处理器是实现时，为了节省面积，会将高位抹去，往往只留 VAddrBits 或者 PAddrBits，设置 mtval 时扩展至 64 bits。NutShell 没有严格实现这一规范，不过目前还没有遇到因为这个问题导致 linux/debian 等程序崩溃的经历，这个设计就一直保留下来了。

作为这一问题的延申，因为处理器流水线中没有保留完整的数据，因此设置 mtval 时需要根据例外类型，进行符号扩展或零扩展，也是麻烦一件呀。

SeddonShen, 2023-04-09:
学习了，谢谢学长，十分感谢！！

closed 2023-04-09 by SeddonShen