OSCPU/NutShell #142: make emu时报错，应该和difftest有关(已经git更新过子模块)

luot-ai, 2023-10-17:
MMIOBase = 0x40000000
MMIOSize = 0x40000000
MemMapBase = 0x0
MemMapRegionBits = 0
MmodeOnly = false
NrExtIntr = 1
ResetVector = 0x80000000
Exception in thread "main" java.util.NoSuchElementException: NOOP_HOME
	at ... ()
	at difftest.dpic.DPIC$.collect(DPIC.scala:197)
	at difftest.DifftestModule$.finish(Difftest.scala:308)
	at sim.SimTop.<init>(NutShellSim.scala:70)
	at top.TopMain$.$anonfun$new$6(TopMain.scala:68)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)
1 targets failed
generator[3.6.0].test.runMain subprocess failed

closed 2023-10-17 by luot-ai

DavidZyy, 2023-11-20:
> MMIOBase = 0x40000000 MMIOSize = 0x40000000 MemMapBase = 0x0 MemMapRegionBits = 0 MmodeOnly = false NrExtIntr = 1 ResetVector = 0x80000000 Exception in thread "main" java.util.NoSuchElementException: NOOP_HOME at ... () at difftest.dpic.DPIC$.collect(DPIC.scala:197) at difftest.DifftestModule$.finish(Difftest.scala:308) at sim.SimTop.(NutShellSim.scala:70) at top.TopMain$.$anonfun$new$6(TopMain.scala:68) at ... () at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace) 1 targets failed generator[3.6.0].test.runMain subprocess failed

How do you solve this problem?

Lemover, 2023-11-20:
` java.util.NoSuchElementException: NOOP_HOME`.
Please set NOOP_HOME to the path of NutShell. cmd like: export NOOP_HOME=/path/to/nutshell