OSCPU/NutShell #146: 在跑Readme中make emu仿真编译时遇到conflicting declaration

youonlylooktwice, 2023-11-20:
我通过git clone --recursive https://github.com/OSCPU/NutShell.git命令将Nutshell以及作为子模块的difftest模块一同下载到本地。
运行make emu指令时，遇到以下问题:
问题是在difftest子模块里路径src/test/csrc/common/golden.h的文件里，函数pte_helper的定义，
与仿真编译的时候，在build/emu-compile/VSimTop__Dpi.h路径文件中的函数pte_helper定义有冲突。
[image]

我的推测是，或许克隆的difftest版本太新了？但我是直接递归克隆的。

下边是整个nutshell目录有调用到pte_helper函数的地方
[image]

SeddonShen, 2024-01-23:
the same problem i have met, did you solve this problem?

huoyin01, 2024-07-18:
同样的问题，请问解决了吗

huoyin01, 2024-07-23:
解决方案：将Verilator升级到5.016以上