OSCPU/NutShell #200: 新版本的Nutshell 按照readme中的教程好像生成不了fpga工程

Taylorsbear720, 2024-06-12:
我在用最新的nutshell生成fpga工程时会报错。
报错原因为， 在执行make之后不会在/build下生成Topmain.v而是在build/rtl下生成TopMain.sv
执行make verilog也是生成TopMain.sv文件，这和之前版本生成的文件不一致，这也导致在进入fpga后生成工程时，会导致说找不到Topmain.v文件。
[image]
[image]

Taylorsbear720, 2024-06-13:
尝试修改了common.tcl 将路径修改为正确的地址加后缀，同时将没有runahead文件注释掉。这样可以生成成功，但是生成的工程没有nutshell.bd，会形成如下的（这里我已经修改过文件的名字）：
[image]
[image]

dzwduan, 2024-07-12:
改一下这里 https://github.com/OSCPU/NutShell/blob/52aeba57513919998a81d18c0b10b1dd1f5ec1ac/src/test/scala/TopMain.scala#L79 
读circt文档

closed 2024-08-26 by poemonsense